drfenixion/freecad.robotcad #17: Difference between the original Cross and this fork

MJavadZallaghi, 2024-07-12:
I was trying to generate URDF using original Cross workbench for FreeCAD, but as there was no tutorial, I can not continue using the tool. Indeed, I'm confused how to configure joints and positioning them using LCF (local coordinate frames) of the parts I've crated.

So I've created this issue to ask:
- What is the difference between your fork and original Cross workbench?
- Is there any tutorial to let me know how I can confidently use it?

drfenixion, 2024-07-13:
To the current moment (13.07.24) all new features from this fork was merged to CROSS. Difference now are "New some icons and icons reordering" and some minor changes.

I am preparing new big update of code generator. It will in new release of OVERCROSS. 

You can use.
Docs: https://github.com/drfenixion/freecad.overcross/wiki
Tutorial - robocar (rus lang, but a lot of screens, just use translator): https://vk.com/@-219386643-overcross-robot-po-shagam-do-ros-2-paketa-mnogo-skrinshoto
In future there will be more tutorials - vk.com/@robotforge-freecad-overcross-uchebka

Also you can use CROSS documentation about "Set placement" tool - https://github.com/galou/freecad.cross/wiki/Command-SetPlacement

MJavadZallaghi, 2024-07-13:
Thanks.
The tutorial worked for me. I was able to move my robot from CAD to Gazebo! Really thank you! 😊🌹🙏🏼

closed 2024-07-13 by MJavadZallaghi