drfenixion/freecad.robotcad #21: Error Freecad Snap Instalation

argolomb, 2024-08-08:
Hi, First, thank you for the great job. 
I'm facing some trouble to launch overcross with Freecad. 
OS: Ubuntu 22.04
ROS: Humble
Freecad 0.21.2

[image]

My suspicious is because Freecad use python 3.8.

drfenixion, 2024-08-08:
If you install RobotCAD manually (looks like you do) you should install dependencies
[https://github.com/drfenixion/freecad.overcross?tab=readme-ov-file#launching-freecad-with-ros](https://github.com/drfenixion/freecad.overcross?tab=readme-ov-file#launching-freecad-with-ros)

Also you can you install and run RobotCAD by 1 command (it will installed in docker container with all depencies)
[https://github.com/drfenixion/freecad.overcross?tab=readme-ov-file#fast-install-and-run](https://github.com/drfenixion/freecad.overcross?tab=readme-ov-file#fast-install-and-run)

drfenixion, 2024-09-21:
Вelved deeper into this issue. 
Yes, this issue related to python verison and minimal required python version for RobotCAD is 3.10.
If you use manual install try to use other FreeCAD build some of them uses python 3.10+ versions.
Or you can just use fast install script. It will install correct FreeCAD version in container.

drfenixion, 2025-02-13:
Use latest stable FreeCAD AppImage (v1.0 at time of comment) with RobotCAD.

closed 2025-02-13 by drfenixion