drfenixion/freecad.robotcad #31: ERROR useradd: user 'root' already exists

Vladimat, 2024-09-30:
[image]

drfenixion, 2024-09-30:
Hello, script is not adapted for sudo. Use without sudo.

Vladimat, 2024-10-01:
without sudo it shows another error:
"ERROR: permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Head "http://%2Fvar%2Frun%2Fdocker.sock/_ping": dial unix /var/run/docker.sock: connect: permission denied"
[image]

drfenixion, 2024-10-01:
Add user to docker group and reboot OS. 
https://github.com/drfenixion/freecad.robotcad/blob/main/docker/README.md#add-your-user-to-docker-group

Vladimat, 2024-10-01:
adding user to the group helped, container created w/o errors.
But Freecad is not launching.

[image]
just logo pops up for few seconds
[image]
it's not critical for me because I installed through Addon Manager

drfenixion, 2024-10-02:
Try to start again by `bash run.bash` and after that ends do `docker container logs ros2_iron_with_freecad` command. It will display logs from container. Copy logs and attache here.

Vladimat, 2024-10-03:
$ docker container logs ros2_iron_with_freecad
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "No module named 'freecad.utils'" occurred in freecad.cross
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 259, in InitApplications
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/vladdy/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/__init__.py", line 7, in <module>
    from freecad.utils import get_python_exe
ModuleNotFoundError: No module named 'freecad.utils'
--------------------------------------------------------------------------------
During initialization the error "No module named 'freecad.utils'" occurred in freecad.cross
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 226, in InitApplications
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/vladdy/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/__init__.py", line 7, in <module>
    from freecad.utils import get_python_exe
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'freecad.utils'
--------------------------------------------------------------------------------
failed to create drawable
failed to create drawable
[1:102:1001/171552.012256:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:102:1001/171552.012269:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:102:1001/171552.012274:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
failed to create drawable
[1:102:1001/171552.032596:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
failed to create drawable
[1:102:1001/171552.052812:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:102:1001/171552.052820:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:102:1001/171552.052822:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
failed to create drawable
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7401f6198520]
#1  /lib/x86_64-linux-gnu/libQt5WebEngineCore.so.5(+0x32c8479) [0x74011f29e479]
QObject::~QObject: Timers cannot be stopped from another thread
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7401f6198520]
#1  0x7401f669cfff in QThreadStorageData::get() const from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x3f
#2  0x7401f6cad5f2 in QOpenGLContext::currentContext() from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x32
#3  0x7401f700f7e8 in QOpenGLFramebufferObjectFormat::QOpenGLFramebufferObjectFormat() from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x38
#4  0x7401ecc0fd51 in QQuickWidget::createFramebufferObject() from /lib/x86_64-linux-gnu/libQt5QuickWidgets.so.5+0x141
#5  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f17c8) [0x7401f68bc7c8]
[... 470 lines cut ...]
--------------------------------------------------------------------------------
failed to create drawable
failed to create drawable
[1:97:1003/183442.380307:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:97:1003/183442.380326:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:97:1003/183442.380331:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
failed to create drawable
[1:97:1003/183442.400918:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
failed to create drawable
[1:97:1003/183442.420586:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[... cut ...]
#3  /lib/x86_64-linux-gnu/libQt5Quick.so.5(+0x3088f1) [0x720d37d8e8f1]
#4  0x720d37d8e933 in QQuickRenderControl::QQuickRenderControl(QObject*) from /lib/x86_64-linux-gnu/libQt5Quick.so.5+0x23
#5  /lib/x86_64-linux-gnu/libQt5QuickWidgets.so.5(+0xb7e1) [0x720d8429a7e1]
#6  /lib/x86_64-linux-gnu/libQt5WebEngineWidgets.so.5(+0x38a65) [0x720d764e9a65]
#7  /lib/x86_64-linux-gnu/libQt5WebEngineWidgets.so.5(+0x24ceb) [0x720d764d5ceb]
#8  /lib/x86_64-linux-gnu/libQt5WebEngineCore.so.5(+0xa775de) [0x720cb0a4d5de]

drfenixion, 2024-10-03:
2 posible reason here: 
1) bug from previous  RoboCAD version (No module named 'freecad.utils')
2) artifacts after sudo (that can lead to Segmentation fault)

Do for fix:
1)  update RobotCAD by `git pull` in RoboCAD directory
2) `chown -R $USER:$USER ~/.local/share/FreeCAD/Mod` (fix root artifacts in RobotCAD mod)
3) `bash run.bash -f` (will recreate container)

If does not helps do again
[https://github.com/drfenixion/freecad.robotcad/issues/31#issuecomment-2388565655](https://github.com/drfenixion/freecad.robotcad/issues/31#issuecomment-2388565655)

Vladimat, 2024-10-04:
result is the same (Freecad logo for few seconds)
probably, it's better to remove all and reinstall
[image]
[image]

drfenixion, 2024-10-04:
It last only Segmentation fault error. Most most likely it is related to wrong permission on files remaining from sudo.

anywhere do
```
sudo chown -R $USER:$USER ~/.local/share/FreeCAD
sudo chmod -R 775 ~/.local/share/FreeCAD
```

in root of robotcad repository do
```
sudo chown -R $USER:$USER .
sudo chmod -R 775 .
```

in docker directory of robotcad repository do
`bash run.bash -f`

If not help attach logs from container.

Vladimat, 2024-10-04:
unfortunately, the same result:
[image]

drfenixion, 2024-10-04:
do this and attach response

`prime-select query`

Vladimat, 2024-10-05:
[image]

drfenixion, 2024-10-05:
"maybe, it's better to remove container and repeat again?" - you do that every time you do `bash run.bash -f` and it does not help.

Open docker/run.bash with any text editor and add # symbol 3 lines as in screenshots.

Before.
[image]

Should be after adding (red dots just a marker).
[image]

Vladimat, 2024-10-05:
[image]
[image]

drfenixion, 2024-10-05:
Try to `DRI_PRIME=0 bash run.bash -f`

if not help 

Remove docker image
`docker image rm osrf/ros:iron-desktop-with_overcross_deps`

And try to `DRI_PRIME=0 bash run.bash -f`

Addition options.
Check your video devices and current OpenGL renderer by `inxi -Gxx` and attach response here. 
"inxi" package my be need to install before.

Vladimat, 2024-10-05:
[image]
[image]
[image]

drfenixion, 2024-10-05:
Just in case if you did not remove image after removing of container:
```
docker rm $(docker ps -aq)
docker image rm osrf/ros:iron-desktop-with_overcross_deps
```
After that `bash run.bash` will rebuild image and container.

About you Graphics. 
I dont see any integraited (in CPU) device and looks like you have only Nvidia or integrated video device deactivated some way.

My current version that errors related to Nvidia card. You cant switch to other device and we will try to use Nvidia.

I also use Nvidia card for container OpenGL renderer. My devices and config:
[image]
I notice you have unloaded "modesetting" but i have it is loaded. Dont sure does it matter or not in this case.

Do next:
1) revert adding "#" symbol (to 3 lines) in bash.run file and save after.
2) Instal Nvidia toolkit for container - [https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt)
Substeps:
[https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#installing-with-apt)
[https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#configuring-docker](https://docs.nvidia.com/datacenter/cloud-native/container-toolkit/latest/install-guide.html#configuring-docker)
3) `bash run.bash -f`

If does not help.
I noticed your OpenGL API based on Mesa library. Articles says that Nvidia proprietary driver can not use Mesa Library. Mesa is opensource and used with Nouveau opensource driver.

I recommed reinstall Nvidia proprietary drivers. It will also install it own library for OpenGL.
install like `apt-get install nvidia-driver`

After reinstall nvidia driver do
`bash run.bash -f`

drfenixion, 2025-03-20:
@Vladimat 
Also was fixed "failed to create drawable" error with `-n` option.

in your case should use
`git pull && cd docker && bash run.bash -bcno`
and after for regular run
`cd docker && bash run.bash -c`