drfenixion/freecad.robotcad #32: Robotcad is installed through AddonManager but not available in Workbench list

Vladimat, 2024-09-30:
[image]
[image]

drfenixion, 2024-09-30:
Check FreeCAD Report View for errors and add them here if exists. 

In theory RobotCAD can conflict with CROSS, try to remove CROSS.

Vladimat, 2024-10-01:
Yes, w/o CROSS it works. Now it generates another error, but it's inherited from CROSS:
"<class 'ModuleNotFoundError'>: No module named 'geometry_msgs'
10:19:44  Running the Python command 'GetPlanningScene' failed:
Traceback (most recent call last):
  File "/home/vladdy/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/ui/command_get_planning_scene.py", line 33, in Activated
    fcgui.addModule('freecad.cross.planning_scene_proxy')

No module named 'geometry_msgs'"

closed 2024-10-01 by Vladimat