drfenixion/freecad.robotcad #34: Python version error

pranavk-2003, 2024-10-05:
I am trying to use Robotcad on freecad windows, although i have python 3.12.7 , Freecad is throwing me the following error
[image]
[image]

what am i doing wrong?

drfenixion, 2024-10-05:
FreeCAD does not use OS python, it uses it own version inside FreeCAD.

You can see FreeCAD Python version by:
Open the Help --> About FreeCAD menu then click on the "Copy to clipboard" button, and paste its content here.

I think your FreeCAD build uses 3.8 Python. Check other FreeCAD builds some of them uses Python 3.10 and 3.10+ that works with RobotCAD.

You can check portaible or system FreeCAD builds for required python version.

pranavk-2003, 2024-10-05:
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/India (en_IN)
[/code]

right here it says Freecad uses py 3.8.10. Yea so now i did got the portable file of freecad which solves my issue . Thank you for a quick response, Appreciate it!

closed 2024-10-05 by pranavk-2003