drfenixion/freecad.robotcad #36: Python version problem with add on install

altineller, 2024-10-13:
Hello,

After succeeding installing rosdeps, I have followed the documentation to configure a custom add on repo, and then tried to install robotcad with the add on manager.

When I click install, I get:

"This addon (or one of its dependencies) require python 3.10 and your system is running 3.8. installation cancelled."

I am on ubuntu 22.04, with ros humble installed. Strange thing is when I run python --version it will return Python 3.10.12 and Python 3.8 is not installed anywhere in my system it seems.

Best Regards,
C.A.

altineller, 2024-10-13:
I went to freecad->about and copied the content below:

[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

I am using the snap version of freecad. The apt version of freecad is older then the required version for robotcad.

Which freecad should I use?

Best Regards,
Can

drfenixion, 2024-10-13:
Try AppImage version. RobotCAD required python 3.10 or more. Some of FreeCAD builds uses 3.10+ python.

FreeCAD uses it`s own python inside. You can not use system python for FreeCAD.

closed 2024-10-14 by altineller