drfenixion/freecad.robotcad #41: Error xhost command not available

mchogan, 2024-11-09:
I got the `xhost` error below after attempting the [fast install and run](https://github.com/drfenixion/freecad.robotcad?tab=readme-ov-file#fast-install-and-run).

```bash
$ bash run.bash
...
run.bash: line 141: xhost: command not found
docker: Error response from daemon: failed to populate volume: error while mounting volume '/var/lib/docker/volumes/9ea232b22c492b2fc7f715ebaa8a2fcce98bfa7eaf6645db6e6bf14656f48bab/_data': failed to mount local volume: mount /home/<user>/<working-directory>/freecad.robotcad/docker/..//docker/ros2_ws//build_data/build:/var/lib/docker/volumes/9ea232b22c492b2fc7f715ebaa8a2fcce98bfa7eaf6645db6e6bf14656f48bab/_data, flags: 0x1000: no such file or directory.
```

***System Config***

- Windows 11
- WSL2 Ubuntu-22.04 distribution from the Microsoft Store
- Windows Docker Desktop 4.35.1 (173168)

***My next steps***

1. I'll try [your instructions here](https://github.com/drfenixion/freecad.robotcad/blob/main/docker/README.md) and see if it helps to install Docker inside the WSL2 Ubuntu-22.04 distribution, rather than use WSL2 with Docker Desktop for Windows.
2. I'll try to find beginner friendly instructions to make xhost GUI apps work in WSL2.

Will report back how it goes... Any advice is appreciated. RobotCAD looks really cool and I'm hoping it will help students with a mechanical engineering and CAD background design and simulate ROS robots.

drfenixion, 2024-11-10:
Instruction how I run RobotCAD at win10, via WSL2
https://m.vk.com/@robotforge-ustanovka-robotcad-na-windows-cherez-wsl-2 (russian language, use translater)
Maybe you should update WSL firstly. 

I only tested successfully run at win10 via WSL2. 
But for real work I use Ubuntu 22.04 directly as dual boot system. 

Other way is to install RobotCAD from add-on manager
https://github.com/drfenixion/freecad.robotcad?tab=readme-ov-file#installation
In that case you will need manually to install dependencies. 

_/home/<user>/<working-directory>/_ is it you replaced user and work directory?

mchogan, 2024-11-10:
Thank you for the note back.

> `mount /home/<user>/<working-directory>/`
> is it you replaced user and work directory?

Yes. When I typed up the issue description I put `<user>` instead of my user name and `<working-directory>` instead of the path to my working directory.

drfenixion, 2025-09-16:
in case of `xhost: command not found`

`apt install x11-xserver-utils`

closed 2025-09-16 by drfenixion