drfenixion/freecad.robotcad #46: The exported URDF is displayed incorrectly

xx-tao, 2024-11-28:
I have created a robot model using RobotCAD from a STEP file, but when I open the exported URDF file in RViz, I receive the error: 'No transform from [l_JT1_CWH_ASM_6_ASM] to [l_DZ1_CWH_ASM_3_ASM]'. How should I go about resolving this issue?
[image]
[image]

System Config

Ubuntu 2404
FreeCAD 1.0
Ros2 Jazzy

drfenixion, 2024-11-28:
You can send me your FreeCAD project. I will check it.

drfenixion, 2024-11-29:
Some basic issues i see in your project.

1) RobotCAD does not support parts with non zero placement nested in other parts in links currently (it will lead to wrong placement of link). Case of DZ1-CWH_ASM_3_ASM.
2) RobotCAD calculate mass and inertia only for first solid in part of link currently. You can check it using collision tool, it will also make collsions only for first solid. 

Resolving: separate solids of parts by different links or fuse solids of same part to one body.

In other words - only one solid body for link.

I did fixes of your solids in parts and placement of base link and tested in ROS2 Iron (current project build with docker via exteded code generator). I did not test it in Ros2 Jazzy. I will make build for Jazzy with docker in external code generator some months later.

https://github.com/user-attachments/assets/f16d9755-920b-41c4-866d-33d1aa937861

drfenixion, 2025-02-13:
Fixed in https://github.com/drfenixion/freecad.robotcad/releases/tag/v6.2.3.

Update your RobotCAD to latest and regenerate project. Also it may be necessary to recalculate inertia and recreate collisions previously.

closed 2025-02-13 by drfenixion