drfenixion/freecad.robotcad #53: how to solve Error：Second and third axes of inertia are undefined.

maqingcai, 2024-12-04:
I tried to convert the .step file into .urdf and run it on rviz. I ran new robot and moved links to see hidden faces on the .step file, but an error message appeared.

[image]

drfenixion, 2024-12-04:
Hello.
"Second and third axes of inertia are undefined"
Looks likes it is not a errors but warnings. It is from "Set placement - fast" tool. It uses LCS as reference on faces with Inertia attachment. if you can correct position joint and link just ignore that warining. Look likes you do placement correct.

By the way, i see unrealistic big mass of your robot.

Here was issue with this model of robot [https://github.com/drfenixion/freecad.robotcad/issues/46#issuecomment-2507515111](https://github.com/drfenixion/freecad.robotcad/issues/46#issuecomment-2507515111)
There are some basic pitfalls that should be treated.

More info about pitfalls:
[https://github.com/drfenixion/freecad.robotcad/issues/48](https://github.com/drfenixion/freecad.robotcad/issues/48)
[https://github.com/drfenixion/freecad.robotcad/issues/49](https://github.com/drfenixion/freecad.robotcad/issues/49)

You can send me your project it.project.developer@gmail.com if this is not enough for resolve. I will check.

I see this robot model second time. Is it some course, college or you just individual learner?

maqingcai, 2024-12-17:
oh,sorry to delay replying to your comments. Question #46 is from the same group as mine. We are solving the problem of converting from .step file to .urdf file, but we encountered some problems in the process, which caused the format of .urdf file to be unable to be visualized in MoveIt. If it is convenient for you, can you help us convert it so that we can check which part has the problem? Thank you! ! ! ! ! !  I will send the .step file to your gmail!!! :)

drfenixion, 2025-02-13:
Fixed in [https://github.com/drfenixion/freecad.robotcad/releases/tag/v6.2.3](https://github.com/drfenixion/freecad.robotcad/releases/tag/v6.2.3).

Update your RobotCAD to latest and regenerate project. Also it may be necessary to recalculate inertia and recreate collisions previously.

closed 2025-02-13 by drfenixion