drfenixion/freecad.robotcad #65: Can not launch RobotCAD with bash run.bash command – Window & container crash

KevinEppacher, 2025-03-10:
When I try to launch RobotCAD with bash run.bash, it opens the logo for a few seconds and in the moment when it should open, the window and the container crashes. I have tried it on Ubuntu 20.04 with ROS Noetic, on Ubuntu 22.04 with ROS2 Humble, but on both computers it has the same behaviour. I also tried forcing a new container with bash run.bash -f. Do I need to install FreeCAD first and then run the skript, or what am I missing. 

Thank you for the contribution.
BR, Kevin

drfenixion, 2025-03-12:
> failed to create drawable
> Program received signal SIGSEGV, Segmentation fault.

There are some posible reasons:

1)
if you have Nvidia card you can try to comment out lines
```
    if [ "$(prime-select query 2> /dev/null)" == 'nvidia' ]; then
        nvidia_options='--gpus all --env NVIDIA_DRIVER_CAPABILITIES=all'
    fi
```
in docker/run.bash

if it will run successfully after then you should install Nvidia docker capabilities and Nvidia proprietary drivers and uncomment these lines. That give you GPU usage in container. Or you can leave thise lines commented out and will use CPU (Gazebo will work to slow)

2)
chown $USER:$USER ~/.local/share/FreeCAD

KevinEppacher, 2025-03-13:
I already use GPU capabilities for other containers. That sould not be the problem. Nevertheless, i tried to uncomment it out, but the window and the container still crashes.

Here are the logs:

kevin@kevin:~/freecad.robotcad/docker$ bash run.bash -f
Force run new container.
Paths: 
$script_dir: /home/kevin/freecad.robotcad/docker
$ws_path: /home/kevin/freecad.robotcad/docker/..//docker/ros2_ws/
$cont_path_ws: /home/kevin//ros2/ros2_ws_with_freecad
$root_of_freecad_robotcad: /home/kevin/freecad.robotcad/docker/../

Command to run in container: freecad
rm: cannot remove '/home/kevin/freecad.robotcad/docker/..//docker/ros2_ws//build_data/install': Permission denied
rm: cannot remove '/home/kevin/freecad.robotcad/docker/..//docker/ros2_ws//build_data/build': Permission denied
rm: cannot remove '/home/kevin/freecad.robotcad/docker/..//docker/ros2_ws//build_data/ros2_system_logs': Permission denied
rm: cannot remove '/home/kevin/freecad.robotcad/docker/..//docker/ros2_ws//build_data/log': Permission denied
Stop and remove old container if present.
non-network local connections being added to access control list
4c937ac0882c7d2ecfb18d5d0b29bf7668d6ee149cf6e6f86fa567f519e5511b
access control enabled, only authorized clients can connect
Ran new container.

FreeCAD mods from host binded to container and will be accessible.

FreeCAD was started inside container. Wait it window load to your host... (about 20 seconds)

The first time, dependencies will be downloaded and installed via pip, which will take longer.

kevin@kevin:~/freecad.robotcad/docker$ docker logs ros2_jazzy_with_freecad
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.

During initialization the error "module 'FreeCAD' has no attribute 'GuiUp'" occurred in freecad.cross
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 259, in InitApplications
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/kevin/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/__init__.py", line 7, in <module>
    import addonmanager_utilities as utils
  File "/usr/share/freecad/Mod/AddonManager/addonmanager_utilities.py", line 44, in <module>
    import addonmanager_freecad_interface as fci
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/freecad/Mod/AddonManager/addonmanager_freecad_interface.py", line 50, in <module>
    if FreeCAD.GuiUp:
       ^^^^^^^^^^^^^
AttributeError: module 'FreeCAD' has no attribute 'GuiUp'
--------------------------------------------------------------------------------
python_exe:  /usr/bin/python3
Collecting urdf_parser_py
  Downloading urdf_parser_py-0.0.4.tar.gz (12 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting lxml (from urdf_parser_py)
  Downloading lxml-5.3.1-cp312-cp312-manylinux_2_28_x86_64.whl.metadata (3.7 kB)
Collecting pyyaml (from urdf_parser_py)
  Downloading PyYAML-6.0.2-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.1 kB)
Downloading lxml-5.3.1-cp312-cp312-manylinux_2_28_x86_64.whl (5.0 MB)
[... 116 lines cut ...]
  #     #   #### ####   ### #     # ####   ##  ##  ##

failed to create drawable
failed to create drawable
[1:108:0313/082805.861452:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:108:0313/082805.861464:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:108:0313/082805.861468:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
failed to create drawable
[1:108:0313/082805.862156:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
failed to create drawable
[1:108:0313/082805.862746:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:108:0313/082805.862750:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:108:0313/082805.862754:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7f03898fb320]
#1  /lib/x86_64-linux-gnu/libQt5WebEngineCore.so.5(+0x31aa589) [0x7f02ddb91589]
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7f03898fb320]
#1  /lib/x86_64-linux-gnu/libQt5WebEngineCore.so.5(+0xa91e85) [0x7f02db478e85]
kevin@kevin:~/freecad.robotcad/docker$ 

I also these commands:
sudo chown -R $USER:$USER ~/.config/FreeCAD
sudo chown -R $USER:$USER ~/.local/share/FreeCAD
sudo chown -R $USER:$USER ~/.FreeCAD

KevinEppacher, 2025-03-13:
ChatGPT suggested to run these commands:

    export LIBGL_ALWAYS_SOFTWARE=1
    export QT_X11_NO_MITSHM=1
    export DISPLAY=$DISPLAY

Normally, I used those commands in my docker compose yaml, thats why I dont know if it is effective. However, I think I got a new error message. Here are the logs. In addition, I asked a friend of mine to also try this repository and he gets the same error massages.

kevin@kevin:~/freecad.robotcad/docker$ LIBGL_ALWAYS_SOFTWARE=1
kevin@kevin:~/freecad.robotcad/docker$ echo $LIBGL_ALWAYS_SOFTWARE
1
kevin@kevin:~/freecad.robotcad/docker$ QT_X11_NO_MITSHM=1
kevin@kevin:~/freecad.robotcad/docker$ DISPLAY=$DISPLAY
kevin@kevin:~/freecad.robotcad/docker$ sudo bash run.bash -f
Force run new container.
Paths: 
$script_dir: /home/kevin/freecad.robotcad/docker
$ws_path: /home/kevin/freecad.robotcad/docker/..//docker/ros2_ws/
$cont_path_ws: /home/root//ros2/ros2_ws_with_freecad
$root_of_freecad_robotcad: /home/kevin/freecad.robotcad/docker/../

Command to run in container: freecad
Stop and remove old container if present.
non-network local connections being added to access control list
a5bb632737651146c760861a545a890b64d673c40fad67fc879e7e6d5b6723b5
access control enabled, only authorized clients can connect
Ran new container.

FreeCAD mods from host binded to container and will be accessible.

FreeCAD was started inside container. Wait it window load to your host... (about 20 seconds)

The first time, dependencies will be downloaded and installed via pip, which will take longer.

kevin@kevin:~/freecad.robotcad/docker$ docker ps
CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES
kevin@kevin:~/freecad.robotcad/docker$ docker logs ros2_jazzy_with_freecad
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.

failed to create drawable
failed to create drawable
[1:108:0313/083756.087157:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:108:0313/083756.087164:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:108:0313/083756.087167:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
failed to create drawable
[1:108:0313/083756.088019:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
failed to create drawable
[1:108:0313/083756.088861:ERROR:gl_surface_glx_qt.cpp(188)] glXCreatePbuffer failed.
[1:108:0313/083756.088863:ERROR:gpu_info_collector.cc(69)] gl::GLContext::CreateOffscreenGLSurface failed
[1:108:0313/083756.088865:ERROR:gpu_info_collector.cc(277)] Could not create surface for info collection.
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7fbc90720320]
#1  /lib/x86_64-linux-gnu/libQt5WebEngineCore.so.5(+0x31aa589) [0x7fbbe37ea589]
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7fbc90720320]
#1  /lib/x86_64-linux-gnu/libc.so.6(__pthread_rwlock_rdlock+0x15) [0x7fbc9077d6d5]
#2  /lib/x86_64-linux-gnu/libcrypto.so.3(CRYPTO_THREAD_read_lock+0xd) [0x7fbc8ba695fd]
#3  /lib/x86_64-linux-gnu/libcrypto.so.3(+0x21ecd4) [0x7fbc8ba57cd4]
#4  /lib/x86_64-linux-gnu/libcrypto.so.3(+0x1fa5c2) [0x7fbc8ba335c2]
#5  /lib/x86_64-linux-gnu/libcrypto.so.3(+0x1faae4) [0x7fbc8ba33ae4]
#6  /lib/x86_64-linux-gnu/libcrypto.so.3(EVP_MD_fetch+0x32) [0x7fbc8ba1c7e2]
#7  /usr/lib/python3.12/lib-dynload/_hashlib.cpython-312-x86_64-linux-gnu.so(+0x6e72) [0x7fbc880e5e72]
#8  /usr/lib/python3.12/lib-dynload/_hashlib.cpython-312-x86_64-linux-gnu.so(+0x71fe) [0x7fbc880e61fe]
#9  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1dc4ac) [0x7fbc920be4ac]
#10  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyObject_Vectorcall+0x5c) [0x7fbc92064afc]
#11  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x40ee) [0x7fbc91fff5ee]
#12  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_FastCallDictTstate+0x60) [0x7fbc92066550]
#13  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_Call_Prepend+0xee) [0x7fbc920667be]
#14  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x20310d) [0x7fbc920e510d]
#15  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1fb6dc) [0x7fbc920dd6dc]
#16  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x8f) [0x7fbc920647af]
#17  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x40ee) [0x7fbc91fff5ee]
#18  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_FastCallDictTstate+0x60) [0x7fbc92066550]
#19  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_Call_Prepend+0xee) [0x7fbc920667be]
#20  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x20310d) [0x7fbc920e510d]
#21  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x1fb6dc) [0x7fbc920dd6dc]
#22  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x8f) [0x7fbc920647af]
#23  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x40ee) [0x7fbc91fff5ee]
#24  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyEval_EvalCode+0x20f) [0x7fbc9217f1ff]
#25  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2f7af8) [0x7fbc921d9af8]
#26  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(+0x2f7c1c) [0x7fbc921d9c1c]
#27  /lib/x86_64-linux-gnu/libpython3.12.so.1.0(PyRun_StringFlags+0x74) [0x7fbc921dd234]
[... cut ...]
#31  0x7fbc932f538f in Gui::Application::activateWorkbench(char const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x8bf
#32  0x7fbc932fdf5f in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x171f
#33  freecad(+0x6037) [0x5626a0f29037]
#34  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7fbc907051ca]
#35  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7fbc9070528b]
#36  freecad(+0x7255) [0x5626a0f2a255]
kevin@kevin:~/freecad.robotcad/docker$

drfenixion, 2025-03-15:
I released new RobotCAD version. Update it (git pull) and use `bash run.bash -bc` (for rebuild image and clear old container logs).

> Command to run in container: freecad
> rm: cannot remove '/home/kevin/freecad.robotcad/docker/..//docker/ros2_ws//build_data/install': Permission denied
> rm: cannot remove [...]

Looks like you did run docker/run.bash with sudo. You should not do that. `bash run.bash` will not work properly with sudo.
After that you should chown to your current user of that dirs.

> failed to create drawable
> Program received signal SIGSEGV, Segmentation fault.
It is reason of crash. I think source of that something with Nvidia container capabilities (check installation) or Nvidia drivers (you an try to reinstall proprietary drivers).
Other way to fix is `prime-select intel` (or other integrad in CPU video). If you have integrated GPU it will change video renderer to it.

Also Segmentation fault can be caused by wrong permission of some used by FreeCAD file.

Let me know if you something helps to fix your crash.

KevinEppacher, 2025-03-16:
Thank you for the changes! Now it launches the freecad wiindow without crashing, but it is black. I get these error messages in the log files: 
WARNING: Target directory /home/kevin/miniconda3/envs/freecad_1_0_312/lib/python3.12/site-packages/__pycache__ already exists. Specify --upgrade to force replacement.
failed to create drawable

kevin@kevin:~/freecad.robotcad/docker$ bash run.bash -bc
Force build new image and run new container are requested.
Clearing old logs is requested.

Paths: 
$script_dir: /home/kevin/freecad.robotcad/docker
$ws_path: /home/kevin/freecad.robotcad/docker/..//docker/ros2_ws/
$cont_path_ws: /home/kevin//ros2/ros2_ws_with_freecad
$root_of_freecad_robotcad: /home/kevin/freecad.robotcad/docker/../

Command to run in container: . /home/$USER/miniconda3/bin/activate && conda activate freecad_1_0_312 && QT_QPA_PLATFORM=xcb freecad
Stop and remove old container if present.
Remove image.
Untagged: osrf/ros:jazzy-desktop-with_overcross_deps
Deleted: sha256:a5f5a9c9b2b2e0a4c48acda44dd944901aafb370d23a8b7606a0493cf58e35b1
Build docker container...
[+] Building 1.6s (36/36) FINISHED                                                                                                                                                                                                                  docker:default
 => [internal] load build definition from dockerfile                                                                                                                                                                                                          0.0s
 => => transferring dockerfile: 6.76kB                                                                                                                                                                                                                        0.0s
 => resolve image config for docker-image://docker.io/docker/dockerfile:1.4                                                                                                                                                                                   0.9s
 => CACHED docker-image://docker.io/docker/dockerfile:1.4@sha256:9ba7531bd80fb0a858632727cf7a112fbfd19b17e94c4e84ced81e24ef1a0dbc                                                                                                                             0.0s
 => [internal] load .dockerignore                                                                                                                                                                                                                             0.0s
 => => transferring context: 186B                                                                                                                                                                                                                             0.0s
 => [internal] load metadata for docker.io/osrf/ros:jazzy-desktop                                                                                                                                                                                             0.6s
 => [internal] load build context                                                                                                                                                                                                                             0.0s
 => => transferring context: 869B                                                                                                                                                                                                                             0.0s
 => [ 1/29] FROM docker.io/osrf/ros:jazzy-desktop@sha256:5ec36860dd0f3ed3800192144f65a3d8eeb7f6b259aa339fbbc1f6ba8d0f44cd                                                                                                                                     0.0s
 => CACHED [ 2/29] RUN echo 'Install system packages'                                                                                                                                                                                                         0.0s
 => CACHED [ 3/29] RUN apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install -y         software-properties-common         mesa-utils         nano         pip         wget &&     pip install debugpy &&     apt-get autoclean &&     apt-ge  0.0s
 => CACHED [ 4/29] RUN echo 'Install FreeCAD Daily'                                                                                                                                                                                                           0.0s
 => CACHED [ 5/29] RUN add-apt-repository ppa:freecad-maintainers/freecad-daily &&     apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install -y         freecad-daily -y &&     apt-get autoclean -y &&     apt-get autoremove -y &&     rm -  0.0s
 => CACHED [ 6/29] RUN deluser --remove-home $(id -un 1000); exit 0                                                                                                                                                                                           0.0s
 => CACHED [ 7/29] RUN echo "* enable custom user: kevin"                                                                                                                                                                                                     0.0s
[... cut ...]
Collecting xmltodict
  Downloading xmltodict-0.14.2-py2.py3-none-any.whl.metadata (8.0 kB)
Downloading xmltodict-0.14.2-py2.py3-none-any.whl (10.0 kB)
Installing collected packages: xmltodict
Successfully installed xmltodict-0.14.2

WARNING: Target directory /home/kevin/miniconda3/envs/freecad_1_0_312/lib/python3.12/site-packages/__pycache__ already exists. Specify --upgrade to force replacement.

Migrating Start Workbench to Start command... done.
failed to create drawable
failed to create drawable

Roblyat, 2025-03-17:
After the pull i started the container with docker compose & launched "freecad-daily" inside the container. It works like this.

version: '3.8'

services:
  freecad-robotcad:
    build:
      context: .
      args:
        UID: ${UID:-1000}
        GID: ${GID:-1000}
        USER: user
        GROUP: user
    image: freecad-robotcad
    container_name: freecad-robotcad-container
    stdin_open: true
    tty: true
    privileged: true
    environment:
      - DISPLAY=${DISPLAY}
      - WAYLAND_DISPLAY=${WAYLAND_DISPLAY}
      - XDG_RUNTIME_DIR=${XDG_RUNTIME_DIR}
      - PULSE_SERVER=${PULSE_SERVER}
      - QT_X11_NO_MITSHM=1
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix:rw
      - ~/.local/share/FreeCAD/Mod:/home/${USER}/.local/share/FreeCAD/Mod
      - ./docker/ros2_ws:/home/${USER}/ros2_ws
    network_mode: bridge
    security_opt:
      - seccomp=unconfined
    shm_size: '512m'
    restart: unless-stopped
    command: freecad-daily

drfenixion, 2025-03-17:
@KevinEppacher, I found one cause of "segmentation fault" and released a new version with a fix. Look at https://github.com/drfenixion/freecad.robotcad/releases/tag/v7.0.1 instructions.

KevinEppacher, 2025-03-18:
I pulled the latest changes, but it still did not work, when launching 

bash run.bash -fco 
bash run.bash -bco

python_exe:  /home/kevin/miniconda3/envs/freecad_1_0_312/bin/python3
Collecting xmltodict
  Downloading xmltodict-0.14.2-py2.py3-none-any.whl.metadata (8.0 kB)
Downloading xmltodict-0.14.2-py2.py3-none-any.whl (10.0 kB)
Installing collected packages: xmltodict
Successfully installed xmltodict-0.14.2

WARNING: Target directory /home/kevin/miniconda3/envs/freecad_1_0_312/lib/python3.12/site-packages/__pycache__ already exists. Specify --upgrade to force replacement.

Migrating Start Workbench to Start command... done.
failed to create drawable
failed to create drawable

With the docker compose file of my colleague @Roblyat, it worked. Perhaps it would be a good contribution in the main branch.

drfenixion, 2025-03-18:
> failed to create drawable

There is no more "Segmentation fault" and that means one probles is fixed.

Looks like later issue is something with video driver or Nvidia container capabilities.

Do you have any clear Ubuntu for try clear install (mb dual boot)?
`cd ~/ && git clone https://github.com/drfenixion/freecad.robotcad.git && cd freecad.robotcad/docker && bash run.bash -c`

Also what GPUs do you have (means discrete and integrated)? In some cases there is only discrete GPU like Nvidia.
For display your graphics config use `inxi -Gxx` command.

KevinEppacher, 2025-03-18:
I use a PC with Dual Boot for Windows 10 and Ubuntu 20.04. My PC has a 4070 Ti RTX Nvidia Graphicscard. I installed 535 Nvidia Driver, I have Cuda 11.3 and I installed the nvidia-container-toolkit, because I run myself gpu accelerated ROS1/2 applications with Docker.

Perhaps I can try to pull the latest changes at the PC at my work. I will keep you updated.

drfenixion, 2025-03-20:
@KevinEppacher 
I did reproduce your error ("failed to create drawable") it happen when i disabled Nvidia container options (i have installed Nvidia container toolkit).
It is posible your enviroment does not meet condition of automatic adding these container options.
I added script option for force add these container options.
Try to use `git pull && cd docker && bash run.bash -fcn`

KevinEppacher, 2025-03-20:
it worked, thank you :) i will soon try to create a description package and test your pipeline

drfenixion, 2025-03-20:
> it worked, thank you :) i will soon try to create a description package and test your pipeline

Glad to read it ) Thank you for point to this issue.

closed 2025-05-12 by drfenixion