drfenixion/freecad.robotcad #71: i do not see the workbench

Dawsatek22, 2025-04-30:
i try to install the workbench on a appimage version of freecad but when i restart the app the workbench does not appear

drfenixion, 2025-04-30:
Hello. Problem is that your FreeCAD AppImage python version (inside it) not 3.12. Maybe it 3.11 or 3.10. Use bash script to run RobotCAD in container (it will install FreeCAD conda build with py3.12 inside) or use conda FreeCAD build with python 3.12 (inside FreeCAD) manually. 

It is can be there is not currently FreeCAD AppImage with python 3.12. 

Also logs you can see in FreeCAD ReportView panel.

drfenixion, 2025-04-30:
I added Conda FreeCAD py3.12 installation instruction to Readme if you prefer to install it manually.
https://github.com/drfenixion/freecad.robotcad/blob/main/README.md#installation

Shaw-Way, 2025-06-12:
I installed via the Addon manager, but when restart FreeCAD, it report following error:
```
During initialization the error "cannot import name 'init_stacks' from 'xacro' (/home/nvidia/miniconda3/envs/freecad/lib/python3.12/site-packages/xacro/__init__.py)" occurred in freecad.cross
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 235, in InitApplications
  File "/home/nvidia/miniconda3/envs/freecad/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 999, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/nvidia/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/init_gui.py", line 4, in <module>
    from .ui import command_assembly_from_urdf  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nvidia/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/ui/command_assembly_from_urdf.py", line 6, in <module>
    from freecad.cross.robot_from_urdf import assembly_from_urdf_path  # FreeCAD's PySide!
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nvidia/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/robot_from_urdf.py", line 15, in <module>
    from freecad.cross.urdf_loader import UrdfLoader
  File "/home/nvidia/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/urdf_loader.py", line 11, in <module>
    from xacro import init_stacks, process_doc, XacroException
ImportError: cannot import name 'init_stacks' from 'xacro' (/home/nvidia/miniconda3/envs/freecad/lib/python3.12/site-packages/xacro/__init__.py)
```

drfenixion, 2025-06-15:
@Shaw-Way, @Dawsatek22, i tested and updated installation by Conda in Installation section.

If your issue is persist try it https://github.com/drfenixion/freecad.robotcad?tab=readme-ov-file#installation
but remove conda  freecad_1_0_312 env  before (`conda deactivate && conda remove --name freecad_1_0_312 --all`) if exist.

Let me know if all work fine for you or not.

Shaw-Way, 2025-06-17:
> [@Shaw-Way](https://github.com/Shaw-Way), looks like it is related to old version of Xacro on pip that installed automatically if you dont have xacro.
> 
> I cant currenly test and fix that on my side because have problem with internet connection to conda package files. If Conda works fine for you. You can try to uninstall Xacro from PIP (where old not suited version) and install Xacro via Conda https://anaconda.org/robostack-jazzy/ros-jazzy-xacro/. Remember install it to RobotCAD namespace (conda activate freecad_1_0_312). Other way is install Xacro from dependency package of RobotCAD https://github.com/drfenixion/freecad.robotcad/tree/main/docker/ros2_ws/src/freecad_cross_rosdep via `rosdep install` and source ROS2 workspace after that.
> 
> Also you can just use fast run script. It will install all required in container https://github.com/drfenixion/freecad.robotcad?tab=readme-ov-file#fast-install-and-run
> 
> [@Shaw-Way](https://github.com/Shaw-Way). let me know if helps or not.

Thank you for your patient response. After adding the following code to `python3.12/site-packages/xacro/__init__.py`, it is now working fine.
```
def init_stacks(file):
    global filestack
    global macrostack
    filestack = [file]
    macrostack = []
```
The function was removed in xacro version 1.13.3.

Dawsatek22, 2025-06-17:
okay that works

Dawsatek22, 2025-06-18:
i got the robot cad working but when i get tho point of set placement i get this: `Running the Python command 'SetCROSSPlacementFast' failed:
Traceback (most recent call last):
  File "/home/d22/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/ui/command_set_placement_fast.py", line 59, in Activated
    set_placement_fast()
  File "/home/d22/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/wb_utils.py", line 1308, in set_placement_fast
    set_placement_by_orienteer(doc, joint, 'Origin', parent_orienteer)
  File "/home/d22/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/wb_utils.py", line 738, in set_placement_by_orienteer
    placement1 = get_placement_of_orienteer(orienteer1, lcs_concentric_reversed = True)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/d22/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/wb_utils.py", line 807, in get_placement_of_orienteer
    lcs, body_lcs_wrapper, placement = make_lcs_at_link_body(orienteer, delete_created_objects, lcs_concentric_reversed)
                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/d22/.local/share/FreeCAD/Mod/freecad.robotcad/freecad/cross/wb_utils.py", line 896, in make_lcs_at_link_body
    original_obj_wrapper.addObject(body_lcs_wrapper)

{'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Body: object is not allowed', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Body: object is not allowed', 'btranslatable': False, 'breported': False}` is that normal? i try to follow this video: `https://youtu.be/NUw6PLPC4x4?si=fvD7yKbJof8DzqVl`

drfenixion, 2025-07-03:
> i got not so much skill with using urdf and i got more success sdf files so that i can use for gazebo simulation.

After export you will have ROS2 package with launcher for Gazebo. It means URDF already integrated in launcher for Gazebo.

Dawsatek22, 2025-08-21:
i am back to to say i tried to set placement but when i attached 2 part and tried to add another part the 3e parts gets added but the first part that was attach did not stay with the other parts attached

Dawsatek22, 2025-08-22:
I use freecadadmin version 0.21.2.And i just mailed the project to you around a hour ago.I hope you received.

drfenixion, 2025-08-26:
@Dawsatek22 , looks like you forgot add collisions and your model in Gazebo go under ground. Make collisions for model links.

drfenixion, 2025-08-26:
@

> is it okay to make youtube videos on my progress and project tutorials or do i need to buy a subscription to do that?

Make videos and tutorials is welcome. You no need buy to subscription. Just dont display code gotten from extended code generator service. If some little fragments of it will accidentally recorded is ok.

drfenixion, 2025-08-30:
@Shaw-Way , where is clip?

drfenixion, 2025-08-31:
@Dawsatek22 , send project to me with that configuration i will check.

Dawsatek22, 2025-08-31:
okay i will try to mail it

Dawsatek22, 2025-08-31:
i sent the project

Dawsatek22, 2025-08-31:
i already sent it

drfenixion, 2025-08-31:
@Dawsatek22 
1) Do not select links when use Set placement - Fast, select only some suited at Real objects.
[image]

2) In some of your binded Real objects you have fc_links with non zero placement inside. Reset to zero placement of that links afte Set placement - Fast. In zero placement it will be in correct position after Set placement - fast.

[image]

[image]

Dawsatek22, 2025-09-01:
it was difficult but i got it somehow working thanks for the help. i go probally refine the design more

drfenixion, 2025-09-01:
@Dawsatek22 , i fixed your issues with selecting redundant objects in project tree and reseting fc_link placement. Update RobotCAD.
https://github.com/drfenixion/freecad.robotcad/releases/tag/v9.3.2

Dawsatek22, 2025-09-04:
ok thanks thats about all for now have a nice day