drfenixion/freecad.robotcad #74: How to imoprt URDF and xacro file?

yes89929, 2025-05-09:
### Environment

- OS: Windows 11 64-bit

- FreeCAD installation: Installed using the Windows installer downloaded from the official website.

- RobotCAD installation: Downloaded the RobotCAD 7.0.2 zip from GitHub and extracted it to "%appdata%/FreeCAD/Mod".

### Steps to Reproduce

1. Run FreeCAD as administrator.

2. Create a new project.

3. Activate the RobotCAD workbench.

4. Observe that the "Import a URDF or xacro file" function is disabled.

[image]

drfenixion, 2025-05-09:
@yes89929
 
Select just a document, not a robot.

[image]

yes89929, 2025-05-12:
Thank you for your reply.
The "Import a URDF or xacro file" function is still disabled.
Do I need to have ROS2 installed to enable this feature?

drfenixion, 2025-05-12:
@yes89929 
Yes, it is check for ROS2. But you can install RobotCAD by [run.bash](https://github.com/drfenixion/freecad.robotcad?tab=readme-ov-file#fast-install-and-run) and ROS2 will be installed automatically in docker container.

yes89929, 2025-05-13:
@drfenixion 
Thank you.

I installed ROS2 and all features are working properly.

In my case, I experienced latency when using Docker, so I installed ROS2 on Windows using conda (mamba).
https://robostack.github.io/GettingStarted.html

Currently, I am using a batch file to run it.
```
@echo off
chcp 65001
:: Check for administrator privileges and request if needed
net session >nul 2>&1
if %errorlevel% neq 0 (
    echo Administrator privileges are required. Requesting permission...
    powershell -Command "Start-Process '%~f0' -Verb runAs"
    exit /b
)

:: Write the actual commands to execute from here
echo Running with administrator privileges!
call mamba activate ros2-humble
start "" "C:\Program Files\FreeCAD 1.0\bin\freecad.exe"
```

I wanted to install FreeCAD 1.0.0 with Python 3.12 using conda as well, but no matter which version I installed, FreeCAD would not run on Windows.

closed 2025-05-13 by yes89929

drfenixion, 2025-05-13:
@yes89929 , if you want to use Win you can to use WSL2 (mb it will require to update WSL) in it. And install Ubuntu 22.04 inside it and further work as in regular Ubuntu.