drfenixion/freecad.robotcad #89: Cannot Set-Placement-Fast correctly

Andy-Leo10, 2025-06-13:
Hello, I did this for Set-Placement-Fast 2 links from a 4-wheel-omni robot
[image]

But, after Set-Placement-Fast, the joint is not generated as expected and also there is not LCS component as the tutorial videos
[image]

Am I doing something wrong? Just in case, I attach my file
https://drive.google.com/file/d/1Db9Z-z_myytRvCXlBER7GoYgsHM78lLu/view?usp=sharing

Thanks for the time and sorry if it is a dumb question

Andy-Leo10, 2025-06-16:
Thanks, I didn't know that using a point from a sketch was wrong.

Now I was able to build the robot. 
[image]

But, I'd like to know how can I place the l_chassis_part TF in the center of the chassis, using robotcad. And also, which are the steps if I want to use a base_footprint (something very common in robots).
[image]

Sorry for asking something else here, but I don't know on which part/channel/forum I can ask these questions.

Andy-Leo10, 2025-06-18:
I did that, thank you so much and congrats for your amazing work!

drfenixion, 2025-06-18:
> congrats for your amazing work!

Thank you.