drfenixion/freecad.robotcad #92: Some links in the exported URDF are invisible

Shaw-Way, 2025-06-25:
These weels is visible in FreeCAD, but invisible in URDF files, and their coliision is correct in simulator.
[image]

[image]

drfenixion, 2025-06-25:
Looks like you sim screen not the same as model version at screen from RobotCAD. It has many differences. Both front wheels present, some floor panel in back present, not motor, no back wheel, etc. Do you sure it is same version?
If so check generated meshes and mesh file names in URDF.

If you cant resolve problem, you can send RobotCAD project to me. I will check.

Shaw-Way, 2025-06-26:
It seems that the model is too complicated, so pybullet is unable to render all the links. Is there any way to simplify meshes when exporting to urdf?

closed 2025-06-27 by Shaw-Way